ydonnelly/ESP8266_fritzing #3: I need more ESP8266 module fzz LOL

hdwong, 2015-05-15:
Hi, I had an ESP8266 ESP-12 wifi module and this project just has 01, 07 model, could you help me, thanks!

greg-ware, 2015-05-16:
ESP-12 has the same layout and pinout as ESP-07, only a slight change in the board antenna.
You could alter ESP-07 slightly.
Would be nice if those boards show up as variants of the same part.

greg-ware, 2015-05-16:
Also, the PCB view is not yet done for ESP-07, so this will not make much of a difference

Killzo, 2015-06-01:
Request: ESP-03

ydonnelly, 2015-06-03:
Hi Killzo, the ESP-03 part has just been added (without the PCB layout for the moment). The pins are slightly misaligned with the grid in the schematic view, I will fix this shortly. As I don't actually have this variant please let me know if there are any inaccuracies.

ydonnelly, 2015-06-03:
The pin alignment seems ok now that I look back at it. Feel free to let me know if there are any other issues.

Killzo, 2015-06-06:
Hi ydonnelly, Thanks for adding ESP-03. However, it it looks like ESP8266-03 WiFi Module.fzpz is just the ESP-01 model with no other variant. Can you verify that you updated the correct file and it's not something I'm doing wrong. Thanks.

ydonnelly, 2015-06-06:
Thanks for pointing that out, I should have double-checked the .fzpz contents before committing. I have recommitted that part, still only breadboard & schematic for the moment.

breagan, 2015-07-27:
Great work!  Thanks for doing this!!

wilmardo, 2016-02-09:
Is there any progress on the PCB of the ESP-03? It would be awesome!

ydonnelly, 2016-02-16:
ESP-03 PCB just added, as always double-check the dimensions with your part before using it!

wilmardo, 2016-02-17:
@ydonnelly Wow that's fast! Many thanks, this makes my day :+1:

Damian-R-Smith, 2016-02-23:
ESP-02 just added

stefanct, 2016-05-17:
I am completely new to Fritzing (used geda/pcb previously) and I am looking for a ESP-12F lib. AFAICT there is none yet... I would prefer someone beat me to creating one :) It is similar to the E variant but with 6 (half)pins opposite to the antenna (http://www.aliexpress.com/item/Esp8266-WiFi-series-of-model-ESP-12-ESP-12F-esp12F-esp12-authenticity-guaranteed/32468324806.html)

ydonnelly, 2016-05-19:
@stefanct Thanks for that find, I will look into merging it.

stefanct, 2016-05-25:
@ydonnelly I am not sure if the 12E is correct though, see my PR #11. As a said I am new to Fritzing (and this will actually by my first and last contribution to its community because I cannot stand it at all), so please take my words and my recently opened pull request with a grain of salt ;)

ydonnelly, 2016-05-26:
I believe the bottom pins in your pull request are flipped, as are GPIO_4 & GPIO_5 (looking at your link and also https://www.adafruit.com/product/2491). I will fix it in the next commit.

stefanct, 2016-05-26:
I cannot confirm that - at all. The points you brought up are exactly what I saw in @thedub2001's E version and there they are wrong. (I read somewhere that the 4/5 swap was actually a bug in the silk layer of the E version but I did not investigate that further).

The correct GPIO order is (from the top/antenna side: 5-4-0). And on the bottom SCLK is near GND while CS is near VCC and both should be the case in the part I pushed AFAICT.
PS: Note that most pictures are mirrored regarding the bottom pins because they are taken from the bottom side of the PCB.

ydonnelly, 2016-05-26:
It appears that the part you pushed had kept the wrong pin positions (SCLK near VCC, GPIO_4 nearer the antenna), so your changes mustn't have copied over to the .fzpz file! I have fixed them in this repo's version and shortened the bottom pads in the PCB view.

ydonnelly, 2016-05-26:
Apologies, I was confusing your and thedub2001's commits. Just accepted the pull request, everything looks perfect.

Damian-R-Smith, 2016-05-27:
I think that adafruit version is slighty different to the one I have here in front of me. The pin layout i had is also the same as what is shown in the "offical" model list docs.

stefanct, 2016-05-27:
https://github.com/esp8266/Arduino/issues/437#issuecomment-112795166 This was what I was talking about earlier. I had another source for this information but found that one when I looked for it again. There is source code to toggle the pin inside in case someone wants an easy test... I'll verify my F version with it this weekend if time allows.

Damian-R-Smith, 2016-05-27:
This is where i get info

http://www.esp8266.com/wiki/doku.php?id=esp8266-module-family