ydonnelly/ESP8266_fritzing #5: ESP 201

uamp, 2015-06-27:
Hi, thank you for the great work! I don't suppose I could persuade you to draw the ESP201 (http://smarpl.com/sites/default/files/images/ESP8266-ESP-201-pin-reference-v01.png).
I have a circuit using this that I want to put on instructables, but lack the ability to draw svg files!

uamp, 2015-06-27:
[image]

ydonnelly, 2015-06-29:
I have just committed an attempt at this board, I did get an error with the PCB view I have to look into but that shouldn't affect the breadboard or schematic views.

closed 2015-06-29 by ydonnelly

ydonnelly, 2015-06-30:
Fixed!

uamp, 2015-06-30:
What a star! Thank you so much.  Hopefully it will benefit lots of others too.